Livox-SDK/livox_camera_lidar_calibration #15: Coordinates not displayed using cornerPhoto

VisionaryMind, 2021-01-20:
Using your sample data as a test, the cornerPhoto launch file opens the first image, but no coordinates are displayed:
[image]

Where can the setting be found for enabling coordinate display?

VisionaryMind, 2021-01-20:
This is a problem with not having OpenCV 4 in Ubuntu 18.04 ROS Melodic uses OpenCV 3.2 by default, so it is necessary to uninstall the Core 3.2 libopencv libraries and manually compile the ROS workspace for OpenCV4.

closed 2021-01-20 by VisionaryMind

rahulsharma11, 2021-09-05:
Hi, I am using ros noetic with opencv4.2

Still my issue is same. I am not able to see the coordinates in bottom of image.

ZYJxxx, 2021-09-23:
> Hi, I am using ros noetic with opencv4.2
> 
> Still my issue is same. I am not able to see the coordinates in bottom of image.

hi, have you solved it?

Baser-Bai, 2021-11-20:
> This is a problem with not having OpenCV 4 in Ubuntu 18.04. ROS Melodic uses OpenCV 3.2 by default, so it is necessary to uninstall the Core 3.2 libopencv libraries and manually compile the ROS workspace for OpenCV4.

Hello,I use opencv4.0  but no cordinate no display.which opencv are you use?

Baser-Bai, 2021-11-20:
> > Hi, I am using ros noetic with opencv4.2
> > Still my issue is same. I am not able to see the coordinates in bottom of image.
> 
> hi, have you solved it?

 hello,have you solved this problem?

1010186, 2022-11-15:
有人解决了吗，is there anyone solved it？who can help children?