Livox-SDK/livox_camera_lidar_calibration #16: getExt1 unable to evaluate residual block with Ceres Solver

VisionaryMind, 2021-01-21:
We have executed your process faithfully, generating eight RGB images with intrinsic / distortion data, picking points on both the images and on the relevant PCD poindclouds. Everything is in order, but the extrinsics cannot be extracted. We are using Ceres 2.0.0 and PCL 1.8.1. Your instructions do not indicate which versions you require.

The following is the console log output:

<details>
<summary>Terminal Log - Extrinsics Capture</summary>

`started roslaunch server http://ubuntu:40985/

SUMMARY
========

PARAMETERS
 * /error_threshold: 12
 * /extrinsic_path: /home/visionarymi...
 * /init_value: [0.0, -1.0, 0.0, ...
 * /input_lidar_path: /home/visionarymi...
 * /input_photo_path: /home/visionarymi...
 * /intrinsic_path: /home/visionarymi...
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    getExt1 (camera_lidar_calibration/getExt1)

auto-starting new master
process[master]: started with pid [21409]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 40fcc524-5b86-11eb-9025-000c29a7eb2a
process[rosout-1]: started with pid [21420]
started core service [/rosout]
process[getExt1-2]: started with pid [21423]
Get the parameters from the launch file
3.604 0.348 0.127 868 453 
3.632 -0.047 0.12 968 450 
3.579 0.355 -0.44 866.199 602.522 
3.591 -0.04 -0.439 970.089 603.662 
3.221 0.954 0.16 680.35 436.59 
3.332 0.584 0.149 797.771 440.533 
3.205 0.967 -0.413 678.422 607.985 
3.309 0.589 -0.424 796.386 602.285 
3.535 0.05 0.133 943 449 
3.449 -0.324 0.126 1045.36 444.462 
3.512 0.057 -0.434 941.998 603.281 
3.472 -0.325 -0.441 1046.23 607.426 
4.471 0.577 0.089 837.117 471.398 
4.61 0.191 0.089 918 472 
4.448 0.58 -0.464 841 593 
4.483 0.202 -0.471 921 590 
4.332 1.24 0.113 697 466 
4.449 0.87 0.097 778.264 469.58 
4.245 1.215 -0.441 693.186 595.975 
0 0 0 778.178 594.362 
4.332 1.24 0.113 837.424 429.42 
4.449 0.87 0.097 960.897 429.78 
4.245 1.215 -0.441 840.903 614.713 
0 0 0 962 608 
2.864 0.723 0.154 724.223 423.595 
3.021 0.371 0.159 850.246 430.882 
2.861 0.724 -0.418 725.17 616.304 
3.011 0.373 -0.413 850.281 615.028 
2.697 0.19 0.166 893.511 414.386 
2.648 -0.194 0.164 1032.19 409.463 
2.676 0.189 -0.405 894.237 616.112 
2.614 -0.198 -0.408 1035.77 618.983 
2.203 0.201 0.184 867.476 381.646 
2.215 -0.175 0.184 1035.6 380.578 
2.174 0.208 -0.38 870.21 632.406 
[... 184 lines cut ...]
xyz are 2.094 0.333 0.192
ErrorU is 19.879 errorV is 89.877
**********************
Data 37 has a error bigger than the threshold
xyz are 2.264 -0.023 0.186
ErrorU is 4.37864 errorV is 87.4853
**********************
Data 38 has a error bigger than the threshold
xyz are 2.075 0.335 -0.38
ErrorU is 9.57661 errorV is 79.2072
**********************
Data 39 has a error bigger than the threshold
xyz are 2.239 -0.021 -0.388
ErrorU is 0.691957 errorV is 77.1114
**********************
u average error is: nan
v average error is: nan

The reprojection error bigger than the threshold, extrinsic result seems not ok
[getExt1-2] process has finished cleanly
log file: /home/visionarymind/.ros/log/40fcc524-5b86-11eb-9025-000c29a7eb2a/getExt1-2*.log
`
</details>

We see this same problem if we use your provided sample data, so the issue is likely not in the data itself.

VisionaryMind, 2021-01-23:
This issue occurs when one of the picked PCD points is 0,0,0. This can happen if a space in between points is clicked and then transferred to the points file.

closed 2021-01-23 by VisionaryMind

Shibaditya99, 2022-04-13:
Hi @VisionaryMind , I don't have any 0,0,0 PCD points, still getting the same error. Do you know from where I can get those initial extrinsic values of LiDAR?