Livox-SDK/livox_camera_lidar_calibration #19: [cameraCalib-2] process has died

CARASO, 2021-02-05:
hjw@hjw-desktop:~$ roslaunch camera_lidar_calibration cameraCalib.launch
... logging to /home/hjw/.ros/log/5d94e990-67a8-11eb-bf97-00e04c68010e/roslaunch-hjw-desktop-8219.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:36375/

SUMMARY
========

PARAMETERS
 * /camera_folder_path: /home/hjw/data/ca...
 * /camera_in_path: /home/hjw/data/ca...
 * /col_number: 6
 * /height: 27
 * /result_path: /home/hjw/data/ca...
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /row_number: 9
 * /width: 27

NODES
  /
    cameraCalib (camera_lidar_calibration/cameraCalib)

auto-starting new master
process[master]: started with pid [8231]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 5d94e990-67a8-11eb-bf97-00e04c68010e
process[rosout-1]: started with pid [8243]
started core service [/rosout]
process[cameraCalib-2]: started with pid [8246]
Get the parameters from the launch file
/home/hjw/data/camera/photos/18.bmp
OpenCV Error: Assertion failed (scn == 3 || scn == 4) in cvtColor, file /build/opencv-XDqSFW/opencv-3.2.0+dfsg/modules/imgproc/src/color.cpp, line 9716
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/opencv-XDqSFW/opencv-3.2.0+dfsg/modules/imgproc/src/color.cpp:9716: error: (-215) scn == 3 || scn == 4 in function cvtColor

[cameraCalib-2] process has died [pid 8246, exit code -6, cmd /home/hjw/camera_lidar_calibration/devel/lib/camera_lidar_calibration/cameraCalib __name:=cameraCalib __log:=/home/hjw/.ros/log/5d94e990-67a8-11eb-bf97-00e04c68010e/cameraCalib-2.log].
log file: /home/hjw/.ros/log/5d94e990-67a8-11eb-bf97-00e04c68010e/cameraCalib-2*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

songjunqianli, 2021-04-29:
in the cameraCalib.cpp file , comment out line 105 : 
`//imshow("Camera Calibration", view_gray);       // 显示图片`

and then , it will be ok . @CARASO @JacoBasil

chenhongjin811, 2022-01-05:
> in the cameraCalib.cpp file , comment out line 105 : `//imshow("Camera Calibration", view_gray); // 显示图片`
> 
> and then , it will be ok . @CARASO @JacoBasil

it did not work

chenhongjin811, 2022-01-05:
> > in the cameraCalib.cpp file , comment out line 105 : `//imshow("Camera Calibration", view_gray); // 显示图片`
> > and then , it will be ok . @CARASO @JacoBasil
> 
> it did not work

my bad, it does work when I change the launch file back

Ricardo-liu12, 2022-02-28:
> 

sorry ,i change cameraCalib.cpp file ,and it did not work 。could you tell me how to change the launch file? thanks a lot @chenhongjin811

Shibaditya99, 2022-04-09:
Opencv has a camera calibration program. you can calibrate camera and save the data from that. I'm not sure of it but to use the next step of camera-lidar calibration  you need only intrinsic and distortion parameters. which you can find from that script.
[Camera_calibration](https://docs.opencv.org/3.4/dc/dbb/tutorial_py_calibration.html)