Livox-SDK/livox_camera_lidar_calibration #24: 中文文档不对

zyxcambridge, 2021-04-02:
https://github.com/Livox-SDK/livox_camera_lidar_calibration/blob/master/doc_resources/README_cn.md

1.3 Download source code and compile
# install camera/lidar calibration package
cd src
git clone https://github.com/Livox-SDK/livox_camera_lidar_calibration.git
cd ..
catkin_make
source devel/setup.bash

1.3 下载源码，编译准备
git clone https://github.com/Livox-SDK/livox_camera_lidar_calibration.git
cd camera_lidar_calibration
catkin_make
source devel/setup.bash

少写一个路径 
cd src

坑坑坑

Livox-SDK, 2021-05-26:
你好~ zyxcambridge:
感谢提出这个问题, 我们稍后会对该readme文档进行更新, 对你带来的困扰十分抱歉

yuyu19970716, 2021-06-04:
你好，请问这个camera/lidar calibration package包是自己建立吗？src也是自己建立吗？怎么我自己建立以后运行roslaunch一直报错，说找不到launch文件呢，应该是路径问题。我也不知道这是怎么回事，应该是我用错了，可以劳烦帮忙解答一下吗，谢谢

yuyu19970716, 2021-06-04:
ERROR: cannot launch node of type [camera_lidar_calibration/cornerPhoto]: Cannot locate node of type [cornerPhoto] in package [camera_lidar_calibration]. Make sure file exists in package path and permission is set to executable (chmod +x)

Livox-SDK, 2021-06-11:
基于你的描述 我怀疑, 
1. 可能改文件没有在launch目录下 
2. 可能没有 source devel/setup.bash

closed 2021-06-16 by Livox-SDK

yangninghua, 2022-01-25:
我也是，一直出现这个错误与