Livox-SDK/livox_camera_lidar_calibration #28: Camera type

Camilochiang, 2021-06-10:
Good day.
Why does the camera need to be a "pinhole imaging model" ? isn't possible to use a normal camera with fix focus length?

Thanks in advance!

Livox-SDK, 2021-06-11:
Hi Camilochiang, 

I think since the optimization logic of the camera is based on pinhole imaging model, That's the reason we used this model here.

closed 2021-06-16 by Livox-SDK

BellosC, 2021-06-22:
Thank you for the answer but can we use a normal camera with fix focus length or not with this package?

Camilochiang, 2021-06-22:
I believe that is the case. if you keep your camera with fix focus (infinite hopefully) It should work as is basically in that case is working as a pin hole camera. I will try and let you know ;)

BellosC, 2021-06-22:
thank you
 @Camilochiang