Livox-SDK/livox_camera_lidar_calibration #32: Coordinates not displayed using cornerPhoto  with opencv4.0

Baser-Bai, 2021-11-20:
no coordinates  when use command
'''roslaunch camera_lidar_calibration cornerPhoto.launch'''

what happened and how solve it?@ @Livox-SDK @

GCaptainNemo, 2021-11-25:
You need to tick the button WITH_QT when compiling opencv library

GCaptainNemo, 2021-11-25:
> no coordinates when use command '''roslaunch camera_lidar_calibration cornerPhoto.launch'''
> 
> what happened and how solve it?@ @Livox-SDK @
You need to tick the button WITH_QT when compiling opencv library

ningdr, 2021-12-29:
OpenCV 4.5.5 + Ubuntu 20.04，I can't get coordinates either.

blackusagi, 2022-01-03:
我也有这个问题，ubuntu18.04下，获取角点数据时，不显示坐标，底栏信息，不能进行下一步，

qulei123, 2022-01-06:
I also encountered this problem. Has anyone solved it?

ailiwei17, 2022-01-08:
Ubuntu18.04
"Can not convert a string to double"

Shibaditya99, 2022-04-09:
same error for me also, i.e. "Can not convert a string to double"

Shibaditya99, 2022-04-13:
Hi there, You can use [this](https://github.com/kaanoguzhan/livox_camera_lidar_calibration) repo to get the coordinates

JasonLrh, 2022-05-24:
> OpenCV 4.5.5 + Ubuntu 20.04，我也无法获取坐标。

1010186, 2022-11-15:
i used the changed code upstairs said,but it still did not work,somebody has other ideas?

ZyueRemi, 2022-12-12:
No coordinate shown on the bottum.