Livox-SDK/livox_camera_lidar_calibration #39: Calculation and U V Error issue.

Minsu-Kim-KNU, 2022-01-10:
Hi, I'm trying to get an extrinsic parameters with LIVOX mid-70 & Intel realsense D435i.

I won all the processes before, but now I'm in 5.2

My result of 
`roslaunch camera_lidar_calibration getExt1.launch`
is,,,

```
xyz are 2.514 0.04 0.801
ErrorU is 2.19356 errorV is 16.2438
```

I think x, y, z value is right, but the error of U and V is compeletly wrong. I think it has calculation error. how can I solve it?

Minsu-Kim-KNU, 2022-01-21:
And, I've got one more issue. all steps are finish, but after run 
`roslaunch camera_lidar_calibration colorLidar.launch`
I can see only black points. any solutions?

BellosC, 2022-02-07:
In my case, some of the livox coordinates that i measured, had an error much greater than the rest of the measurements.
When i deleted those measurements from the corner_photo.txt and corner_lidar.txt and ran again "roslaunch camera_lidar_calibration getExt1.launch", the error was acceptable. After that i had a good colorized pointcloud result.

>you can find all the values and errors, after executing "roslaunch camera_lidar_calibration getExt1.launch". If you scroll up in your terminal you can see the error of each livox-photo corner that you measured. If you find something that doesn't seems right follow the procedure that i did and perhaps you will be lucky.

For your second question, i believe that because you failed to calibrate the livox with the camera correctly as you said, the only thing that is available for you to see for now is the x,y,z position info and not the r,g,b coloring info. So it makes perfect sense to see only black points. After you solve the problem of your 1st question i believe you will see your colorized point-cloud.

Good luck.

Minsu-Kim-KNU, 2022-02-12:
I have already solved this problem. thanks.

CallmeAG, 2022-03-30:
> I have already solved this problem. thanks.
Hi，I have met the same problem when i use the data given by author, May I know how you solve the problem at the end?

Minsu-Kim-KNU, 2022-03-30:
Of course.

First, I calibrate the camrea again really properly I think this step is the most important..
As we know, nowdays cameras are calibrated well when it comes for factory.
but, I recommend that you must do it again.

And then, follow this step correctly. this is second important thing.

and, If you have camera parameters(intrinsic, distortion) from first step, You must run only getExt1. not getEx2.

After clear all these steps, you can see the beautifully colorized point cloud. good luck

CallmeAG, 2022-03-31:
Thank you so much for your suggestion.
I did exactly what the document says again with the data given by the link. But the result was still unacceptable.
I checked the output in the terminal and it said " The reprojection error bigger than the threshold, extrinsic result seems not ok."
So I doubt that It's somethings wrong with the data.  I'm going to try the code with my own camera and lidar next.
thanks again for replying me.

springslx, 2022-09-30:
Hi, I get the same errord with the provided sample data. Did you solve the problem?
`
intrin:
952.339 0 793.155
0 949.096 255.112
0 0 1

Use the extrinsic result to reproject the data
Data 10 has a error bigger than the threshold
xyz are 4.773 -1.285 -0.381
ErrorU is 11.1892 errorV is 6.6026

Data 11 has a error bigger than the threshold
xyz are 4.75 -1.285 1.139
ErrorU is 3.69858 errorV is 11.7171

Data 19 has a error bigger than the threshold
xyz are 4.731 0.333 1.054
ErrorU is 0.107852 errorV is 12.0369

Data 27 has a error bigger than the threshold
xyz are 4.495 0.06 0.997
ErrorU is 1.48431 errorV is 20.8934

Data 28 has a error bigger than the threshold
xyz are 4.296 0.43 0.966
ErrorU is 4.52526 errorV is 24.4341

Data 29 has a error bigger than the threshold
xyz are 4.292 0.437 -0.369
ErrorU is 4.78589 errorV is 8.26999

Data 31 has a error bigger than the threshold
xyz are 4.383 -0.559 0.991
ErrorU is 0.533192 errorV is 18.0158

u average error is: 3.03319
v average error is: 6.02012`

[image]

springslx, 2022-09-30:
@Minsu-Kim-KNU Hi, I get the same errord with the provided sample data. Did you solve the problem?
`
intrin:
952.339 0 793.155
0 949.096 255.112
0 0 1

Use the extrinsic result to reproject the data
Data 10 has a error bigger than the threshold
xyz are 4.773 -1.285 -0.381
ErrorU is 11.1892 errorV is 6.6026

Data 11 has a error bigger than the threshold
xyz are 4.75 -1.285 1.139
ErrorU is 3.69858 errorV is 11.7171

Data 19 has a error bigger than the threshold
xyz are 4.731 0.333 1.054
ErrorU is 0.107852 errorV is 12.0369

Data 27 has a error bigger than the threshold
xyz are 4.495 0.06 0.997
ErrorU is 1.48431 errorV is 20.8934

Data 28 has a error bigger than the threshold
xyz are 4.296 0.43 0.966
ErrorU is 4.52526 errorV is 24.4341

Data 29 has a error bigger than the threshold
xyz are 4.292 0.437 -0.369
ErrorU is 4.78589 errorV is 8.26999

Data 31 has a error bigger than the threshold
xyz are 4.383 -0.559 0.991
ErrorU is 0.533192 errorV is 18.0158

u average error is: 3.03319
v average error is: 6.02012`

[image]

Minsu-Kim-KNU, 2022-09-30:
@springslx 

Hi, I think you U and V error get quite better.
However, it's quite wired since your intrinsic parameters are 793 and 255.
I think you need to calibration your camera then, try to get extrinsic parameter with Livox again.
Also, getEXt1 is only for extrinsic camera parameter. so, keep in mind do not use getExt2 since it also change your intrinsic parameter.

I recommand that calibrate the camera with its own.

I hope you get to the goal. Thanks.

springslx, 2022-09-30:
> @springslx
>
> Hi, I think you U and V error get quite better. However, it's quite wired since your intrinsic parameters are 793 and 255. I think you need to calibration your camera then, try to get [...]

Thanks for your reply. I will try it with my own data.

phrdsgznb, 2022-10-25:
Excuse me,I have a question about the initial value in getExt1.launch,I noticed in the Readme that "If the cost is still relatively large at the end of the iteration (for example, 10 to the power of 4), it may be a problem with the initial value setting in the launch file, the result is only a local optimal solution, then the initial value needs to be reset",but I want to ask for advice about 
how to obtain the initial value of the position relationship between camera and lidar