Livox-SDK/livox_camera_lidar_calibration #44: FoV of camera and LiDAR

PatilVrush, 2022-06-16:
To ensure that the field of view of the photo and the field of LiDAR point cloud are consistent, the photo dimension needs to be modified according to the FOV(field of view) of LiDAR, the Horizon ’s FOV is 81.7 (horizontal) x 25.1 (vertical), so the photos used in this case have a dimension of 1520 (width) x 568 (height) pixels. Can someone please help me understand how does this work for a custom camera lens of different FoV and resolution.

ckgpeace, 2022-10-21:
@PatilVrush : were you able to figure this out? I am using DepthAI camera of resolution 1080*1920. I am able to run the whole project but colorLidar.launch repeats the left side of the photo. Attaching the color.Lidar.launch result. I think this might be due to FOV mismatch. Please help me out here.
[image]

zza1617, 2022-10-28:
I had the same problem@

idiotcv, 2022-11-23:
me to. no body reslove this problem