Livox-SDK/livox_camera_lidar_calibration #45: extrinsic result seems not ok--after run roslaunch camera_lidar_calibration getExt1.launch

ajiang17, 2022-07-22:
Hi,
   I‘m using camera_lidar_calibration getExt1.launch to get lidar to camera  extrinsics .  I'm using 10 pairs of pics and pcds ,  total 40points.
   Logs are pasted below.  Although CONVERGENCE, the project error is much bigger than threshhold.  
   Please Help me,    what is wrong ? 
   
---------------Logs:
iter      cost      cost_change  |gradient|   |step|    tr_ratio  tr_radius  ls_iter  iter_time  total_time
   0  3.263738e+05    0.00e+00    6.46e+05   0.00e+00   0.00e+00  1.00e+04        0    7.46e-03    5.42e-02
   1  1.841992e+05    1.42e+05    1.67e+05   1.47e+00   9.54e-01  3.00e+04        1    1.46e-02    6.89e-02
   2  1.768749e+05    7.32e+03    5.10e+03   2.50e-01   1.00e+00  9.00e+04        1    4.70e-05    6.90e-02
   3  1.768657e+05    9.27e+00    9.36e+00   1.09e-02   1.00e+00  2.70e+05        1    7.41e-05    6.91e-02
Ceres Solver Report: Iterations: 4, Initial cost: 3.263738e+05, Final cost: 1.768657e+05, Termination: CONVERGENCE
  -0.721294   -0.692535   0.0122162
-0.00404571  -0.0134298   -0.999907
   0.692631   -0.721267   0.0068757
   0.264165
-0.00771116
   -1.74526
Use the extrinsic result to reproject the data
Data 0 has a error bigger than the threshold
xyz are 5.83209 -6.24977 0.447996
ErrorU is 13.851 errorV is 9.07203

.......

**********************
Data 36 has a error bigger than the threshold
xyz are 4.55385 -4.30335 0.334322
ErrorU is 27.3254 errorV is 14.5054
**********************
Data 38 has a error bigger than the threshold
xyz are 4.07123 -4.82789 -0.658529
ErrorU is 19.669 errorV is 11.8685
**********************
Data 39 has a error bigger than the threshold
xyz are 4.55385 -4.30335 -0.658529
ErrorU is 20.7527 errorV is 20.7066
**********************
u average error is: 27.4879
v average error is: 34.0989

The reprojection error bigger than the threshold, extrinsic result seems not ok

chychiang, 2022-08-05:
Getting the same error with the provided sample data. 
```
u average error is: 13.7469
v average error is: 14.7411
```

springslx, 2022-09-30:
@ajiang17 @chychiang Hi, I get the same errord with the provided sample data. Did you solve the problem?
u average error is: 3.03319
v average error is: 6.02012

springslx, 2022-09-30:
@ajiang17 @chychiang Hi, I get the same errord with the provided sample data. Did you solve the problem?
`
intrin: 
952.339       0 793.155
      0 949.096 255.112
      0       0       1

Use the extrinsic result to reproject the data
Data 10 has a error bigger than the threshold
xyz are 4.773 -1.285 -0.381
ErrorU is 11.1892 errorV is 6.6026
**********************
Data 11 has a error bigger than the threshold
xyz are 4.75 -1.285 1.139
ErrorU is 3.69858 errorV is 11.7171
**********************
Data 19 has a error bigger than the threshold
xyz are 4.731 0.333 1.054
ErrorU is 0.107852 errorV is 12.0369
**********************
Data 27 has a error bigger than the threshold
xyz are 4.495 0.06 0.997
ErrorU is 1.48431 errorV is 20.8934
**********************
Data 28 has a error bigger than the threshold
xyz are 4.296 0.43 0.966
ErrorU is 4.52526 errorV is 24.4341
**********************
Data 29 has a error bigger than the threshold
xyz are 4.292 0.437 -0.369
ErrorU is 4.78589 errorV is 8.26999
**********************
Data 31 has a error bigger than the threshold
xyz are 4.383 -0.559 0.991
ErrorU is 0.533192 errorV is 18.0158
**********************
u average error is: 3.03319
v average error is: 6.02012`

springslx, 2022-09-30:
[image]