Livox-SDK/livox_camera_lidar_calibration #52: 激光雷达数据保存为bag包时，使用roslaunch livox_ros_driver livox_lidar_msg.launch，然后怎么保存为customMsg格式

1010186, 2022-12-15:


1010186, 2022-12-15:
@Livox-SDK

1010186, 2022-12-15:
[ INFO] [1671087467.176839337]: Init lds lidar success!
[ WARN] [1671087467.186709118]: Shutdown request received.
[ WARN] [1671087467.186750218]: Reason given for shutdown: [[/livox_lidar_publisher] Reason: new node registered with same name]
.
.
.
[ERROR] [1671087471.099274355]: Client [/rviz] wants topic /livox/lidar to have datatype/md5sum [sensor_msgs/PointCloud2/1158d486dd51d683ce2f1be655c3c181], but our version has [livox_ros_driver/CustomMsg/e4d6829bdfe657cb6c21a746c86b21a6]. Dropping connection.

lilith235, 2024-04-09:
roslaunch livox_ros_driver livox_lidar_msg.launch 录制的点云格式为customMsg，roslaunch livox_ros_driver livox_lidar.launch录制的点云格式为PointCloud2，用后面这个录